nofeletru/UsbAsp-flash #1: Wake up from deep power down needed

igor-m, 2017-10-27:
Hi, thanks for developing the AsProgrammer and the usbasp SPI modification!!

I see an issue when trying to program a SPI flash memory "on-the-fly", where the SPI flash was put into "Deep Power Down" by the command 0x89 already.
Your programmer shall first issue the "Release from Power-down" command, code 0xAB, to the SPI flash in order to wake up the flash from deep power down.
Currently my FPGA reads the SPI flash memory and puts it into the deep power down afterwards. After that I cannot program the SPI flash unless I switch off/on the power while I force the FPGA not to read the SPI flash memory at the power on.
So the very first command you have to send to the SPI flash memory is 0xAB. That command has no impact on memories lacking this feature. I use a standard 25xx SPI memory.

nofeletru, 2017-10-29:
This is the corrected version
Please confirm that it works
https://cloud.mail.ru/public/M1Lo/i9TPE4GVH

igor-m, 2017-10-31:
Ok, it works.   Thanks!

closed 2017-10-31 by nofeletru