nofeletru/UsbAsp-flash #4: Programming speed

igor-m, 2018-03-10:
I do program a 104kB bitstream into a 4MB SPI flash via the modded usbasp. Works fine :)
The bitstream is located first 104kB.
It takes 30secs to erase and 30secs to program the flash. In total 1minute.
Is that because it re-flashes the entire 4MB chip, or, because there is a delay inserted somewhere, or, because the usbasp's communication is slow?

nofeletru, 2018-03-10:
usbasp speed in spi mode is about 7kB/s to 15kB/s (you may check it in benchmark menu)
Result speed is lower because need of transfer not only data but some instructions for memory communication. (and verify if checked)
So yeah - usbasp is slow =)

If file contain only data to write(with no trailing FF) - erase whole chip - write only what in file.
Erasing 8Mbit flash is taking about 5 seconds(we just send 1 command and wait).

igor-m, 2018-03-10:
The 4MBytes (32Mbits) takes 30secs here.
The binary is only 104kBytes large, so I need to erase 2 blocks (64kBytes is the block size with the 0xD8 Sector Erase command) and program it. That may take (7.8kBytes/sec here) 1sec for erase and 20secs for write. That would be a great speed up :)
How to do it?

nofeletru, 2018-03-11:
Look at blockerase.pas script.

igor-m, 2018-03-11:
But how to tell ASprogrammer "do use my Erase script" when I press the erase icon in the main window?
I've modified the erase script such it erases only 3 Blocks and it takes 2 seconds. I have to press the Execute button in the Script editor window (Script = ERASE) however.

nofeletru, 2018-03-11:
In chiplist.xml you need to set script file 
<M45PE16 id="204015" page="256" size="2097152" script="blockerase.pas"/>
And now program will override action with specified section in script file

{$read} read button
{$write} write button...
{$erase}
{$verify} 
{$unlock}

igor-m, 2018-03-11:
Ok, it works!
So for example in a "my_commands.pas" I can have all the read, write, erase ... actions together?

nofeletru, 2018-03-12:
Yes