nofeletru/UsbAsp-flash #6: USBasp trouble reading Macronix 25V4035

davidcie, 2018-09-07:
Not sure if you take bug reports here or on the forum but just in case... I have the above chip on an HDD PCB I'm working on. If I use my CH341A programmer with AsProgrammer v1.4.0, the IC is read correctly. However, the same AsProgrammer v1.4.0 reads all 0s if I use USBasp flashed to the latest firmware supplied with AsProgrammer. Tried a few different connection combinations too. 

I can explain in more detail how I hook up USBasp if needed. If you need some oscilloscope traces to debug this, I think I could try taking them. Let me know what you need _or_ if this is a hardware limitation and USBasp for some reason cannot and will never be able to correctly read a 25V4035.

nofeletru, 2018-09-14:
usbasp have some hardware problems.
First of all is the contact pad far away from microcontroller.
Better practise is use cap near programmable ic and use cap(0,1uf) from CS to ground.

davidcie, 2018-10-03:
Thanks, will try the next time I get one of these for programming. However, I also discovered that USBasp seems to use 5V signals even though I have 3.3V selected and so can be dangerous for some low voltage chips. I therefore tend to go for the true 3.3V CH341A in these cases.

closed 2018-10-03 by davidcie