nofeletru/UsbAsp-flash #11: Possible to support MX35LFxGE4AB SPI Nand ?

protocold, 2019-11-07:
Datasheet is here: https://www.macronix.com/Lists/Datasheet/Attachments/7666/MX35LF2GE4AB,%203V,%202Gb,%20v1.7.pdf

Is it possible at all to add support for this chip?

nofeletru, 2019-11-07:
It uses SPI, so that is possible. The difference is only in the commands.

protocold, 2019-11-07:
would you kindly help to have a look? Thank you so much!

engmarcos26, 2020-01-07:
nofeletru, can you make a video showing how you do to add a new chip, how do you program that?

nofeletru, 2020-02-20:
Sorry but there is no time to work on project right now.
Easy way to add new spi chip is to write all logic on script language(pascal too =) ). So you not need to recompile anything.
Most chips like W25QXX have the same commands and all you need is to add parameters in chiplist.xml

engmarcos26, 2020-02-21:
Alexander, Could you send me a tutorial in English how did you integrate Ci
programmer with the paschal environment? I would like to understand how to
do this type of application!

*Atenciosamente*,

*Eng. Marcos Domingos *
Engenheiro Mecatrônico / Eletrotécnico / Instrumentista Industrial
(82) 996925113
www.linkedin.com/in/eng-marcos-domingos-42aa20117

engmarcos26, 2020-05-04:
Здравствуйте. Я использую переводчик Google, поэтому я не знаю, является ли
русский перевод правильным. Но что я хотел бы знать, так это то, что вы
создали программу для программирования регистратора eeprom CH301A?
Я хотел бы, чтобы вы объяснили шаги по созданию программы для управления
этим оборудованием ch301A. Спасибо

engmarcos26, 2020-05-04:
Sorry. Correct is Programmer CH341A.

Em seg., 4 de mai. de 2020 às 03:25, Eng. Marcos Domingos <
eng.marcos26@gmail.com> escreveu:

> Здравствуйте. Я использую переводчик Google, поэтому я не знаю, является
> ли русский перевод правильным. Но что я хотел бы знать, так это то, что вы
> создали программу для программирования [...]