nofeletru/UsbAsp-flash #17: Compiling

Mattiwatti, 2020-02-20:
Hi, thanks for your work on this.

Would it be possible to have in the `README` some short steps and required software needed to compile the software? I've never used Pascal before so it isn't obvious to me which variant of Pascal is being used, or even what the main project file is (`AsProgrammer.lpi`? `AsProgrammer.lps`?). Thanks.

Mattiwatti, 2020-02-20:
Update: I managed to compile the sources using [Lazarus 2.0.6](https://www.lazarus-ide.org/) (note: must be the 32 bit version, or at least, I couldn't get the Win64 -> Win32 cross compiler to work). The project depends on the `richmemopackage` and `mphexeditorlaz` packages which are included in the repository as ZIPs.

I did have two errors compiling `richmemopackage`:
```
win32richmemoproc.pas(258,10) Error: Identifier not found "UnicodeEnabledOS"
win32richmemoproc.pas(492,8) Error: Identifier not found "UTF8Length"
```
I worked around this by implementing these myself. But maybe I overlooked something.

It would still be nice if some 'official' compilation steps could be added to the `README`. But my executable works now so that's good.

nofeletru, 2020-02-20:
richmemopackage is outdated
https://sourceforge.net/p/lazarus-ccr/svn/HEAD/tree/components/richmemo/

nofeletru, 2020-02-20:
Also look at test branch https://github.com/nofeletru/UsbAsp-flash/tree/test

closed 2020-06-01 by nofeletru

mrdc, 2023-01-06:
> Hi, thanks for your work on this.
>
> Would it be possible to have in the `README` some short steps and required software needed to compile the software? I've never used Pascal before so it isn't [...]

You need AsProgrammer.lpr, I've builded it without any issues using the latest Lazarus 2.2.4 64bit. First time using Lazarus :)