nofeletru/UsbAsp-flash #39: Support GD25LQ20

vinibali, 2021-09-14:
Hi!
Is there any way to support this chip? GD25Q20 doesn't seem to erase or write the chip.
I saw a commit for GD25LQ devices, but this one is not working at the moment.
@JuliusSenkus, which device did you check?
[Added to chip list GIGADEVICE GD25LQ064C_1.8V](https://github.com/nofeletru/UsbAsp-flash/commit/2d7ecf0cab2439dc72fa9be90c7480472cee0bb3)

JuliusSenkus, 2021-09-20:
As you mentioned, tested that same device I added, but assumed it was locked (note: if you lock that chip, it will be hard locked as I do remember per chip's documentation), also tested it on broken tablet BIOS chip, so take that with a kg of salt.

nofeletru, 2021-09-25:
It looks like the difference is only in the supply voltage.

closed 2021-09-25 by nofeletru