nofeletru/UsbAsp-flash #43: Can you add W25Q256JVEQ? can you explain how to add it step by step pls!! ;)

bananamarcia, 2021-11-01:
[image]

nofeletru, 2021-11-02:
Download new chiplist.xml
https://github.com/nofeletru/UsbAsp-flash/blob/master/chiplist.xml

bananamarcia, 2021-11-02:
There are two W25Q256JVEQ with two diffrent id... can you pls tell me which is mine?

nofeletru, 2021-11-02:
> There are two W25Q256JVEQ with two diffrent id... can you pls tell me which is mine?

W25Q256JV IQI /IN /JQ 184019h 
W25Q256JV M * /JM * 187019h 
it does not matter  since it uses same commands

bananamarcia, 2021-11-02:
okay, thanks...

bananamarcia, 2021-11-02:
could you help me with another matter??
I have an HP Elitebook 830 g5, and my ch341a programmer doesn't read my bios W25Q256JVEQ... WHAT SHOULD I DO?

nofeletru, 2021-11-02:
Did you desolder chip? If program do not  read chip ID then this some wireing problem.

bananamarcia, 2021-11-03:
I didn't desolder the chip. 
I will try to explain the situation, maybe maybe I was wrong in some steps. I just disconnected the battery and the CMOS from my hp elitebook 830 g5. Then without desolding the BIOS chip I connected it to the programmer with the tweezers provided in the kit, after I installed the programmer drivers.
You said it could be a wiring problem, about what, the programmer or the tweezers?

nofeletru, 2021-11-03:
It highly recommended to desolder chip. 
No guarantee that you can program ic on main board without desoldering it.

closed 2021-11-17 by nofeletru