nofeletru/UsbAsp-flash #54: Support for MX25L25673G

WhatADirtyOldGit, 2021-12-13:
Hi,

I tried to look at the chiplist.xml but don't understand all the parameters required to add a new chip - what needs to go in the "id" field and how do you determine the "size"?

e.g. 
<MX25U25673G_1.8V id="C22539" page="256" size="33554432"/>

I'm trying to add an MX25L25673G M2I-08G

https://www.macronix.com/en-us/products/NOR-Flash/Serial-NOR-Flash/Pages/spec.aspx?p=MX25L25673G&m=Serial%20NOR%20Flash&n=PM2323

https://www.macronix.com/Lists/Datasheet/Attachments/7989/MX25L25673G,%203V,%20256Mb,%20v1.7.pdf

Any help would be greatly appreciated.

Sumebody, 2021-12-13:
try
<MX25L25673G id="C22019" page="256" size="33554432"/>

Sumebody, 2021-12-13:
try
MX25L25673G id="C22019" page="256" size="33554432"

closed 2021-12-16 by nofeletru