nofeletru/UsbAsp-flash #110: Add Puva P25q80H chiplist ?

lazye53, 2024-02-02:
I am using an amd rx 5700xt graphics card and unfortunately the model is not available in the chip list, can you add it because I cannot write the bios properly.

[Puva]
Model: P25Q80H
Id: 881F01D
Voltage: 3.3V

[image]

arieslaptop, 2024-02-14:
<P25Q80H id="881F01D" page="256" size="1048576"/>

requeijaum, 2024-07-31:
Seems that 6600XT also use this chip.

------

```
PS C:\Users\ricar\amdvbflash_v5_techpowerup> .\amdvbflash.exe -ai
AMD IFWI Flasher Tool Version 5.0.567.0-External. Copyright┬® 2020-2022 Advanced Micro Devices, Inc. All rights reserved.

Detecting AMD GPU/APU. Please wait...

Adapter 0 (0x0000:07:00.0)
    ASIC Family                   Navi23
    Flash Type                    P25Q80H
    Product Name                  NAVI23 XT A0 8GB 500e/500m 0.9V/1.35V
    Bios Config File              66XTS23B.bin
    Bios P/N                      xxx-xxx-xxx
    Bios Version                  020.003.000.030.000000
    Bios Date                     10/22/22 22:34

Image 0 -- Offset 0x0000
__________________
  ROM header contents:
    Signature                     0xaa55
    PCIR Offset                   0x0364
  PCI Data Structure:
    Signature                     PCIR☻
    Vendor ID                     0x1002
    Device ID                     0x73ff
    PCI Revision                  0x00
    Image Size                    0xaa00
    Code Revision                 0x1403
    Indicator                     0x00     (Not Last Image)
    Code Type                     0x00     (Intel IA-32, PC-AT compatible)
    Legacy BIOS File Name         66XTS23B.bin
    Legacy BIOS Part Number       xxx-xxx-xxx
    Legacy BIOS Build Number      612782
    Legacy BIOS Change List       2394362
    Binary BIOS Identifier (IDTF) 0xc4f92de2

Image 1 -- Offset 0xaa00
__________________
    Signature                     0xaa55
    PCIR Offset                   0x001c
  PCI Data Structure:
    Signature                     PCIR☻
    Vendor ID                     0x1002
    Device ID                     0x73ff
    PCI Revision                  0x00
    Image Size                    0x9600
    Code Revision                 0x0000
    Indicator                     0x00     (Not Last Image)
    Code Type                     0x03     (EFI)
    Image Size                    0x00000ef1
    Compression Type              0x00     (Compressed)
    MachineType                   0x8664   (X64 Machine Type)
    Subsystem                     0x0b     (EFI Boot Service Driver)
    EFI Image Offset              0x64     (abs: 0xaa64)
  UEFI IDs from EFI_PCI_EXPANSION_ROM_HEADER
    UEFI Source Revision          GOP AMD REV: 000.000.000.020.003
    UEFI Source Change List       2394362
    UEFI BIOS Build Number        612782
  GOP Driver Info:
    Machine Type Code             0x8664   (AMD64 (K8))
    Authenticode Signed           true
    GOP Image Size                0xdd00   (55KB)
    Driver Name                   AMD GOP X64 Release Driver Rev.3.1.0.Mar 14 2021.23:17:45
    UEFI Major Revision           0x03
    UEFI Minor Revision           0x01
    UEFI Fix Revision             0x00
    Legacy Major Version          0x00
    Legacy Minor Version          0x00
    GOP AMD Build
    GOP AMD CL
    GOP BIOS Identifier (IDTF)    0xdeadbeef
    Driver Carries legacy tables  false

Image 2 -- Offset 0x14000
__________________
    Signature                     0xaa55
    PCIR Offset                   0x001c
  PCI Data Structure:
    Signature                     PCIR☻
    Vendor ID                     0x1002
    Device ID                     0x73ff
    PCI Revision                  0x00
    Image Size                    0x9c00
    Code Revision                 0x0000
    Indicator                     0x80     (Last Image)
    Code Type                     0x03     (EFI)
    Image Size                    0x00000ef1
    Compression Type              0x80     (Compressed)
    MachineType                   0xaa64   (X64 Machine Type)
    Subsystem                     0x0b     (EFI Boot Service Driver)
    EFI Image Offset              0x64     (abs: 0x14064)
  UEFI IDs from EFI_PCI_EXPANSION_ROM_HEADER
    UEFI Source Revision          GOP AMD REV: 000.000.000.020.003
    UEFI Source Change List       2394362
    UEFI BIOS Build Number        612782
  GOP Driver Info:
    Machine Type Code             0xaa64   (ARM64)
    Authenticode Signed           true
    GOP Image Size                0xf8e0   (62KB)
    Driver Name                   AMD GOP ARM64 Release Driver, Rev. 3.1.0.Mar 14 2021.23:19:12
    UEFI Major Revision           0x00
    UEFI Minor Revision           0x00
    UEFI Fix Revision             0x00
    Legacy Major Version          0x00
    Legacy Minor Version          0x00
    GOP AMD Build
    GOP AMD CL
    GOP BIOS Identifier (IDTF)    0x0000ff
    Driver Carries legacy tables  false

PS C:\Users\ricar\amdvbflash_v5_techpowerup>
```

closed 2024-10-30 by nofeletru