nofeletru/UsbAsp-flash #127: missing IC cfeon s40-75Gip

JasonB73, 2024-10-22:
Hello, Trying to read a Seagate 1000 drive controller board so I can move the BIOS to my new board.
Can't read from old drive which is 1.8v.
The newer board is 3.3v which I can read just fine.

cfeon s40-75Gip, Are there any other devices that will crossover.
I found a crossover for my newer board and it worked.

(25q40bw -Winbond) W25Q40BW, This one worked to read as (W25Q40BV)

Thanks

nofeletru, 2024-10-24:
W25Q40B are same size as EN25S40.
Winbond W25Q40B[W] id 1.8v and W25Q40B[V] is 3.3v
Eon EN25[S]40 id 1.8v EN25[Q]40 is 3.3v

JasonB73, 2024-10-24:
Chip wouldn't read because it was still on the bad board.
Once I pulled it I was able to read and move the BIOS to the new board.
Got the drive working again with the data intact..
Thanks again for the info and this software project..

closed 2024-10-24 by JasonB73